greatscottgadgets/luna #18: design a r0.3

ktemkin, 2020-05-07:
Some potential changes for r0.3:

- [ ] Re-spec 3V3 regulator to reduce ripple; or at the least, remove its bypass capacitor.
- [ ] Improve grounding / thermal connections for the USB PHYs.
- [ ] Potentially switch between the active-high and active-low versions of our load switches / switch the load configuration to reduce leakage.

ktemkin, 2020-05-15:
@mossmann assigning you `r0.3`-related tasks, for now. Feel free to re-assign. :)

mossmann, 2020-05-21:
Work in progress is in this branch: https://github.com/mossmann/luna/tree/r0.3

mossmann, 2020-05-21:
I've removed bypass capacitor C52 from the +3V3 regulator at least as a temporary measure. I will likely replace the regulator entirely after some testing.

ktemkin, 2020-08-07:
Added one more bullet:

- Connect PROGRAM_N to a nearby 3V3 FPGA I/O to allow for easier Debug Controller-less operation.

This will allow the FPGA to trigger its own reconfiguration.

mossmann, 2020-12-08:
I resolved the load switch issue primarily by reducing the resistance of the pull-up and pull-down resistors on the load switch enable pins.

PR #82 (2021-01-19): LUNA hardware r0.3
closes #18 

This would merge cleanly if I had remembered to pull 6162fc8 into my branch back in July.
closed 2021-01-19 by ktemkin