greatscottgadgets/luna #32: Incorrect bInterfaceProtocol value for ACM Serial

BracketMaster, 2020-07-20:
I tried the ACMSerial example on my TinyFPGA.
I'm using a Mac.
Lsusb shows the USB device, but the MacOS kernel
neglects to mount it as an ACM device.

I found that Luna sets the ``bInterfaceProtocol`` to ``2``.
https://github.com/greatscottgadgets/luna/blob/master/luna/gateware/usb/devices/acm.py#L107

TinyFPGA USB-Serial core which mounts on MacOS sets
that value to ``0``.
https://github.com/davidthings/tinyfpga_bx_usbserial/blob/master/usb/usb_serial_ctrl_ep.v#L371

According the the 1999 ADC [USB specification](https://cscott.net/usb_dev/data/devclass/usbcdc11.pdf)
Section 4.4 page 30,, 0x02 is not implemented and 0x00 is the correct value.

[image]

The ACM example still works on Linux since the Linux ACM driver
is more forgiving. But this is still an issue none the less.

I can open a pull request to fix this.

ktemkin, 2020-07-27:
Thanks for reporting this -- it looks a lot like I typo'd and inserted a `0x02` when I meant `0x01`. Testing stuff out on macOS now and will report back. :)

BracketMaster, 2020-07-27:
Haha. No probs.
Also, MacOS wouldn't allow me to write to the device with screen.
I was however able to write to the device directly with PyUSB...

Which brings me to my next comment. Perhaps it might make sense to 
include a PyUSB serial demo in the same folder as the ACMSerial demo
python code.

ktemkin, 2020-07-27:
MacOS seems happy with the technically-correct value of `0x01`. About to test on Windows.

> Also, MacOS wouldn't allow me to write to the device with screen.
> I was however able to write to the device directly with PyUSB...

There was another bug that I think was triggering this behavior (macOS was interleaving control request transactions with its IN tokens; and a bug in handling that caused the IN endpoint to get stalled). That's fixed in 087ad9c2376658ab7e82d46f4f010e77ff55d3c1, and `dterm` seems happy to work with the terminal.

I'll try screen after I try Windows to verify. :)

commit 34cbbb83 (2020-07-27): gateware.devices: ACM: fix typo in interface number

Correct from 0x02 (reserved) to 0x01 (AT modem).

Fixes #32.
closed 2020-07-27 by ktemkin