greatscottgadgets/luna #36: Building eptri demo fails

kgugala, 2020-07-27:
When I tried to build the eptri demo I got bunch of missing type errors - `error: unknown type name 'uint8_t'` 

adding `#include <stdint.h>` to `eptri_example.c` solves the above, but still i get bunch of implicit declarations warnings:

```
make                                                                                                                                                  
riscv64-unknown-elf-gcc -march=rv32i -mabi=ilp32 -g -Os -Wall -Werror -Tsoc.ld -Triscv_standalone.ld  -nostdlib start.S eptri_example.c -o eptri_example.elf
eptri_example.c: In function 'print_char':
eptri_example.c:119:9: error: implicit declaration of function 'uart_tx_rdy_read' [-Werror=implicit-function-declaration]
  while(!uart_tx_rdy_read());
         ^~~~~~~~~~~~~~~~
eptri_example.c:120:2: error: implicit declaration of function 'uart_tx_data_write' [-Werror=implicit-function-declaration]
  uart_tx_data_write(c);
  ^~~~~~~~~~~~~~~~~~
eptri_example.c: In function 'read_setup_request':
eptri_example.c:167:10: error: implicit declaration of function 'setup_have_read' [-Werror=implicit-function-declaration]
   while(!setup_have_read());
          ^~~~~~~~~~~~~~~
eptri_example.c:170:18: error: implicit declaration of function 'setup_data_read' [-Werror=implicit-function-declaration]
   uint8_t byte = setup_data_read();
^~~~~~~~~~~~~~~
eptri_example.c: In function 'send_packet':                                           
eptri_example.c:187:2: error: implicit declaration of function 'in_ep_reset_write' [-Werror=implicit-function-declaration]
  in_ep_reset_write(1);
  ^~~~~~~~~~~~~~~~~
eptri_example.c:191:3: error: implicit declaration of function 'in_ep_data_write' [-Werror=implicit-function-declaration]
   in_ep_data_write(*buffer);
   ^~~~~~~~~~~~~~~~ 
( ... )
```

disabling `-Werror` ends up in undefined reference linker errors. Looks like some code is missing here (or is not generated before the example is built)

ktemkin, 2020-07-27:
Sounds like something changed in one of the upstreams; probably.  I'll check it out soon.

kgugala, 2020-09-16:
@ktemkin did you have a chance to take a look at this?

ktemkin, 2020-09-23:
I haven't come back around to looking at it; mostly because I've been laser focused on other things. I can take a look sometime this or next week, though. Poke me if I don't. :)

ktemkin, 2020-09-24:
Just tested with the environment I already had installed without issue; it's using:

```
minerva => 0.1
nmigen-soc => 0.1.dev30+g425692a
lambdasoc =>  0.1.dev21+g95a97de
```

I'm guessing upgrading one of those python packages is going to break things -- trying that now. :)

ktemkin, 2020-09-24:
Hmm; even after upgrading everything to its latest, I'm still not having issues:

```
minerva => 0.1 [but now installed from master]
nmigen-soc => 0.1.dev43+gecfad4d
lambdasoc =>  0.1.dev25+g7381b47
riscv64-unknown-elf-gcc => 9.2.0
riscv64-unknown-elf-newlib => 3.3.0
```

I'm particularly confused in that you're winding up with no `stdint.h` include, as `simplesoc` should always include that in the header it generates:

https://github.com/greatscottgadgets/luna/blob/master/luna/gateware/soc/simplesoc.py#L420

Some things that might be helpful in debugging:

1. What steps are you running to build? The simplest way is probably just to run `make program` in the demo folder.
2. Can you attach the `resources.h` and `soc.ld` that are generated by these build steps?

kgugala, 2020-11-04:
looks like `nmigen-stdio` was missing in my setup. The build process was actually showing that in the log, but did not stop on missing dependency. This ended up with empty `resources.h` and in the end the reported error. Adding `nmigen-stdio` to requirements.txt solves the problem

antoinevg, 2024-01-04:
eptri example code has moved to the Cynthion repo: https://github.com/greatscottgadgets/cynthion/

closed 2024-01-04 by antoinevg