greatscottgadgets/luna #65: Analyzer: Implement (PS)RAM buffering

ktemkin, 2020-08-27:
Currently, the analyzer uses block ram for buffering, which leads it to overrun very, very quickly. We should instead buffer the data in (PS)RAM, so we can have a good buffer depth.

MartinLoeper, 2021-06-14:
@ktemkin I am currently trying to read data from a HOST <-> Printer connection with LUNA instead of the USBProxy project.

Everything works well so far until the board goes into overrun state and I have to re-build and upload the applet again.
I also noticed that I have to wait with plugging the printer in until I start to read from the analyzer. Otherwise, the buffer overruns immediately after the applet starts to run.
Is it possible to somehow reset the analyzer state?

I read your comments about the [DRAM buffer support](https://github.com/greatscottgadgets/luna/blob/5d983e1ed82f6a7f14a0509ba9541a0c72c43f7f/luna/gateware/usb/analyzer.py#L20) and I really hope this feature gets attention over the next months.
That would allow us to test a LUNA device in the field, reading data from a POS <-> printer connection.

As always, thanks for the amazing work!

martinling, 2024-02-13:
The analyzer gateware has moved to the Cynthion repository; this issue is now tracked at https://github.com/greatscottgadgets/cynthion/issues/51.

closed 2024-02-13 by martinling