greatscottgadgets/luna #70: Compiling saturn-v fails with arm-none-eabi-gcc 10.2.0

cyber-murmel, 2020-09-03:
I tried to compile saturn-v and it fails with `multiple definition of 'USB_dtype';`
`deps/usb/usb_standard.h` already has `#pragma once` as a include guard.
Adding a `#ifndef` include guard didn't help.

I'm also not getting smart from the error message.
`dfu_block_offset` and `usb_setup` aren't functions. 
`usb_cb_reset`, `usb_reset` contains no keyword from the `USB_dtype`enum.

<details>
<summary>Full Build Log</summary>

```bash
marble@x270 ~/repos/luna/firmware/saturn-v (git)-[master] % git pull
warning: Pulling without specifying how to reconcile divergent branches is
discouraged. You can squelch this message by running one of the following
commands sometime before your next pull:

  git config pull.rebase false  # merge (the default strategy)
  git config pull.rebase true   # rebase
  git config pull.ff only       # fast-forward only

You can replace "git config" with "git config --global" to set a default
preference for all repositories. You can also pass --rebase, --no-rebase,
or --ff-only on the command line to override the configured default per
invocation.

Fetching submodule firmware/lib/tinyusb
Fetching submodule firmware/lib/tinyusb/hw/mcu/microchip/samd/asf4
Fetching submodule firmware/lib/tinyusb/hw/mcu/nordic/nrfx
Fetching submodule firmware/lib/tinyusb/hw/mcu/nxp
Fetching submodule firmware/lib/tinyusb/hw/mcu/sony/cxd56/spresense-exported-sdk
Fetching submodule firmware/lib/tinyusb/hw/mcu/st/st_driver
Fetching submodule firmware/lib/tinyusb/hw/mcu/ti
Fetching submodule firmware/lib/tinyusb/tools/uf2
Fetching submodule firmware/lib/tinyusb/tools/uf2/hidapi
Fetching submodule firmware/saturn-v/deps/sam0
Fetching submodule firmware/saturn-v/deps/usb
Already up to date.

marble@x270 ~/repos/luna/firmware/saturn-v (git)-[master] % git rev-parse HEAD
e72bf698a321f77d05dd430f5ea3c725141beb26

marble@x270 ~/repos/luna/firmware/saturn-v (git)-[master] % make clean
rm -f common/startup_samd21.o main.o usb.o common/clock.o deps/usb/class/dfu/dfu.o deps/usb/samd/usb_samd.o deps/usb/usb_requests.o

marble@x270 ~/repos/luna/firmware/saturn-v (git)-[master] % make
arm-none-eabi-gcc -Wall --std=gnu99 -Os -g3 -flto -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -mcpu=cortex-m0plus -mthumb -D __SAMD21G18A__ -I . -D USB_PRODUCT_ID=0x7551 -D USB_VENDOR_ID=0x1209 -D USB_MANUFACTURER_STR='"Great Scott Gadgets"' -D USB_PRODUCT_STR='"LUNA Saturn-V RCM Bootloader"' -D COPYRIGHT_NOTE='"Visit https://githib.com/opendime/DAFU"'  -Ideps/usb -Ideps/sam0/cmsis -Ideps/sam0/include -Ideps/sam0/cmsis/samd21/include -Ideps/sam0/cmsis/samd21/source   -c -o common/startup_samd21.o common/startup_samd21.c
arm-none-eabi-gcc -Wall --std=gnu99 -Os -g3 -flto -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -mcpu=cortex-m0plus -mthumb -D __SAMD21G18A__ -I . -D USB_PRODUCT_ID=0x7551 -D USB_VENDOR_ID=0x1209 -D USB_MANUFACTURER_STR='"Great Scott Gadgets"' -D USB_PRODUCT_STR='"LUNA Saturn-V RCM Bootloader"' -D COPYRIGHT_NOTE='"Visit https://githib.com/opendime/DAFU"'  -Ideps/usb -Ideps/sam0/cmsis -Ideps/sam0/include -Ideps/sam0/cmsis/samd21/include -Ideps/sam0/cmsis/samd21/source   -c -o main.o main.c
arm-none-eabi-gcc -Wall --std=gnu99 -Os -g3 -flto -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -mcpu=cortex-m0plus -mthumb -D __SAMD21G18A__ -I . -D USB_PRODUCT_ID=0x7551 -D USB_VENDOR_ID=0x1209 -D USB_MANUFACTURER_STR='"Great Scott Gadgets"' -D USB_PRODUCT_STR='"LUNA Saturn-V RCM Bootloader"' -D COPYRIGHT_NOTE='"Visit https://githib.com/opendime/DAFU"'  -Ideps/usb -Ideps/sam0/cmsis -Ideps/sam0/include -Ideps/sam0/cmsis/samd21/include -Ideps/sam0/cmsis/samd21/source   -c -o usb.o usb.c
arm-none-eabi-gcc -Wall --std=gnu99 -Os -g3 -flto -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -mcpu=cortex-m0plus -mthumb -D __SAMD21G18A__ -I . -D USB_PRODUCT_ID=0x7551 -D USB_VENDOR_ID=0x1209 -D USB_MANUFACTURER_STR='"Great Scott Gadgets"' -D USB_PRODUCT_STR='"LUNA Saturn-V RCM Bootloader"' -D COPYRIGHT_NOTE='"Visit https://githib.com/opendime/DAFU"'  -Ideps/usb -Ideps/sam0/cmsis -Ideps/sam0/include -Ideps/sam0/cmsis/samd21/include -Ideps/sam0/cmsis/samd21/source   -c -o common/clock.o common/clock.c
arm-none-eabi-gcc -Wall --std=gnu99 -Os -g3 -flto -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -mcpu=cortex-m0plus -mthumb -D __SAMD21G18A__ -I . -D USB_PRODUCT_ID=0x7551 -D USB_VENDOR_ID=0x1209 -D USB_MANUFACTURER_STR='"Great Scott Gadgets"' -D USB_PRODUCT_STR='"LUNA Saturn-V RCM Bootloader"' -D COPYRIGHT_NOTE='"Visit https://githib.com/opendime/DAFU"'  -Ideps/usb -Ideps/sam0/cmsis -Ideps/sam0/include -Ideps/sam0/cmsis/samd21/include -Ideps/sam0/cmsis/samd21/source   -c -o deps/usb/class/dfu/dfu.o deps/usb/class/dfu/dfu.c
arm-none-eabi-gcc -Wall --std=gnu99 -Os -g3 -flto -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -mcpu=cortex-m0plus -mthumb -D __SAMD21G18A__ -I . -D USB_PRODUCT_ID=0x7551 -D USB_VENDOR_ID=0x1209 -D USB_MANUFACTURER_STR='"Great Scott Gadgets"' -D USB_PRODUCT_STR='"LUNA Saturn-V RCM Bootloader"' -D COPYRIGHT_NOTE='"Visit https://githib.com/opendime/DAFU"'  -Ideps/usb -Ideps/sam0/cmsis -Ideps/sam0/include -Ideps/sam0/cmsis/samd21/include -Ideps/sam0/cmsis/samd21/source   -c -o deps/usb/samd/usb_samd.o deps/usb/samd/usb_samd.c
arm-none-eabi-gcc -Wall --std=gnu99 -Os -g3 -flto -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -mcpu=cortex-m0plus -mthumb -D __SAMD21G18A__ -I . -D USB_PRODUCT_ID=0x7551 -D USB_VENDOR_ID=0x1209 -D USB_MANUFACTURER_STR='"Great Scott Gadgets"' -D USB_PRODUCT_STR='"LUNA Saturn-V RCM Bootloader"' -D COPYRIGHT_NOTE='"Visit https://githib.com/opendime/DAFU"'  -Ideps/usb -Ideps/sam0/cmsis -Ideps/sam0/include -Ideps/sam0/cmsis/samd21/include -Ideps/sam0/cmsis/samd21/source   -c -o deps/usb/usb_requests.o deps/usb/usb_requests.c
arm-none-eabi-gcc -Wall --std=gnu99 -Os -g3 -flto -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -mcpu=cortex-m0plus -mthumb -D __SAMD21G18A__ -I . -D USB_PRODUCT_ID=0x7551 -D USB_VENDOR_ID=0x1209 -D USB_MANUFACTURER_STR='"Great Scott Gadgets"' -D USB_PRODUCT_STR='"LUNA Saturn-V RCM Bootloader"' -D COPYRIGHT_NOTE='"Visit https://githib.com/opendime/DAFU"'  -Ideps/usb -Ideps/sam0/cmsis -Ideps/sam0/include -Ideps/sam0/cmsis/samd21/include -Ideps/sam0/cmsis/samd21/source -o bootloader.elf -flto -Wl,--gc-sections --specs=nano.specs -Wl,-Tlink-script.ld common/startup_samd21.o main.o usb.o common/clock.o deps/usb/class/dfu/dfu.o deps/usb/samd/usb_samd.o deps/usb/usb_requests.o
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: usb.o (symbol from plugin): in function `usb_cb_reset':
(.text+0x0): multiple definition of `USB_dtype'; main.o (symbol from plugin):(.text+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: deps/usb/class/dfu/dfu.o (symbol from plugin): in function `dfu_block_offset':
(.text+0x0): multiple definition of `USB_dtype'; main.o (symbol from plugin):(.text+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: deps/usb/samd/usb_samd.o (symbol from plugin): in function `usb_reset':
(.text+0x0): multiple definition of `USB_dtype'; main.o (symbol from plugin):(.text+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: deps/usb/usb_requests.o (symbol from plugin): in function `usb_setup':
(.text+0x0): multiple definition of `USB_dtype'; main.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:81: bootloader.elf] Error 1
```
</details>

FFY00, 2020-09-03:
GCC 10 changed this behavior and no longer allows multiple definitions by default. It seems like USB_dtype is defined in two different places.

You need to do something similar to what I did here, https://github.com/tendra/tendra/pull/27.

I can have a look later if I have time.

cyber-murmel, 2020-09-04:
Declaring the enum extern seems to do the trick.
```diff
diff --git a/usb_standard.h b/usb_standard.h
index 52bb69f..b22f459 100644
--- a/usb_standard.h
+++ b/usb_standard.h
@@ -55,7 +55,7 @@ enum {
 	USB_FEATURE_TestMode = 0x02,
 };
 
-enum {
+extern enum {
 	USB_DTYPE_Device = 0x01,
 	USB_DTYPE_Configuration = 0x02,
 	USB_DTYPE_String = 0x03,
````

ktemkin, 2021-02-03:
Should be fixed as of the latest master. :)

closed 2021-02-03 by ktemkin