greatscottgadgets/luna #81: (Fomu) examples/soc/: "nmigen.build.res.ResourceError: Resource uart#0 does not exist"

tcal-x, 2020-12-24:
This may well be a user error, but this is what I attempted:

With LUNA_PLATFORM=luna.gateware.platform.fomu:FomuPVT

In luna/examples/soc/hello/

```
$ python3 -m hello_world_soc
module: luna.gateware.platform.fomu  name: FomuPVT
INFO    | __init__    | Building and uploading gateware to attached Fomu PVT/Production...
INFO    | simplesoc   | Physical address allocations:
INFO    | simplesoc   |     00000000-00001000: <luna.gateware.soc.memory.WishboneROM object at 0x7f026399c5b0>
INFO    | simplesoc   |     00001000-00002000: <luna.gateware.soc.memory.WishboneRAM object at 0x7f026399c910>
INFO    | simplesoc   |     00002000-00002004: (rec uart_divisor r_data r_stb w_data w_stb)
INFO    | simplesoc   |     00002004-00002008: (rec uart_rx_data r_data r_stb)
INFO    | simplesoc   |     00002008-0000200c: (rec uart_rx_rdy r_data r_stb)
INFO    | simplesoc   |     0000200c-00002010: (rec uart_rx_err r_data r_stb)
INFO    | simplesoc   |     00002010-00002014: (rec uart_tx_data w_data w_stb)
INFO    | simplesoc   |     00002014-00002018: (rec uart_tx_rdy r_data r_stb)
INFO    | simplesoc   |     00002020-00002024: (rec uart_ev_status r_data r_stb)
INFO    | simplesoc   |     00002024-00002028: (rec uart_ev_pending r_data r_stb w_data w_stb)
INFO    | simplesoc   |     00002028-0000202c: (rec uart_ev_enable r_data r_stb w_data w_stb)
INFO    | simplesoc   |     00002040-00002044: (rec timer_reload r_data r_stb w_data w_stb)
INFO    | simplesoc   |     00002044-00002048: (rec timer_en r_data r_stb w_data w_stb)
INFO    | simplesoc   |     00002048-0000204c: (rec timer_ctr r_data r_stb w_data w_stb)
INFO    | simplesoc   |     00002050-00002054: (rec timer_ev_status r_data r_stb)
INFO    | simplesoc   |     00002054-00002058: (rec timer_ev_pending r_data r_stb w_data w_stb)
INFO    | simplesoc   |     00002058-0000205c: (rec timer_ev_enable r_data r_stb w_data w_stb)
INFO    | simplesoc   |     00002060-00002064: (rec leds_output r_data r_stb w_data w_stb)
INFO    | simplesoc   | 
INFO    | simplesoc   | IRQ allocations:
INFO    | simplesoc   |     0: uart
INFO    | simplesoc   |     1: timer
INFO    | simplesoc   | 
INFO    | simplesoc   | 
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/tim/greatscottgadgets/luna/examples/soc/hello/hello_world_soc.py", line 103, in <module>
    top_level_cli(design, cli_soc=design.soc)
  File "/home/tim/greatscottgadgets/luna/luna/__init__.py", line 140, in top_level_cli
    products = platform.build(fragment,
  File "/home/tim/icebreaker/icebreaker-nmigen-examples/env/src/nmigen/nmigen/build/plat.py", line 95, in build
    plan = self.prepare(elaboratable, name, **kwargs)
  File "/home/tim/icebreaker/icebreaker-nmigen-examples/env/src/nmigen/nmigen/build/plat.py", line 135, in prepare
    fragment = Fragment.get(elaboratable, self)
  File "/home/tim/icebreaker/icebreaker-nmigen-examples/env/src/nmigen/nmigen/hdl/ir.py", line 39, in get
    obj = obj.elaborate(platform)
  File "/home/tim/greatscottgadgets/luna/examples/soc/hello/hello_world_soc.py", line 92, in elaborate
    uart_io = platform.request("uart", 0)
  File "/home/tim/icebreaker/icebreaker-nmigen-examples/env/src/nmigen/nmigen/build/res.py", line 62, in request
    resource = self.lookup(name, number)
  File "/home/tim/icebreaker/icebreaker-nmigen-examples/env/src/nmigen/nmigen/build/res.py", line 57, in lookup
    raise ResourceError("Resource {}#{} does not exist"
nmigen.build.res.ResourceError: Resource uart#0 does not exist
```

ktemkin, 2021-01-06:
Oh! I think when we spoke earlier I didn't get that you were using some of the SoC examples. _Those_ don't currently work on `fomu`; because the SoC example chosen implements a hardware UART, rather than a virtual one.

If you're trying to use this with a _virtual_ serial device, that'd require implementing a simple SoC peripheral that, in turn, interfaces with LUNA's simple ACM virtual-serial, as here:

https://github.com/greatscottgadgets/luna/blob/master/examples/usb/acm_serial.py

ktemkin, 2021-01-06:
A (more complex) example of a peripheral is provided here:

https://github.com/greatscottgadgets/luna/blob/master/luna/gateware/usb/usb2/interfaces/eptri.py

You'd theoretically need to wrap LUNA's `USBSerialDevice` with a Peripheral, and then add e.g. a CSR (register) you can use to send data. I think that'd be a super interesting example -- feel free to open a feature request. :)

ktemkin, 2021-01-06:
I'm closing this since it's not technically a bug; I'd love to have this referenced in a new feature request, though, if you're willing to write one. :)

closed 2021-01-06 by ktemkin

mithro, 2021-01-07:
@ktemkin - I think @tcal-x wants to use the LUNA core to handle the USB serial side and LiteX to provide the CSR peripherals. I /think/ the best way is to set up LUNA to provide a byte-FIFO like interface similar to what the FTDI ICs provide, then connect that to the LiteX byte UART peripheral which would normally talk to something like the FTDI IC? Does that make sense?

mithro, 2021-01-07:
My poor attempt at doing a diagram is shown at https://docs.google.com/drawings/d/1JMEMoRK5CP7TDy4U9-tuYUZqEjzRmwVNJWENpQxgR9g/edit
[image]

tcal-x, 2021-01-07:
Thanks @ktemkin , yes we chatted a bit on discord (1bitsquared).   So yes, things are working as expected, not a bug.  I think I did have success with the acm\_serial.py design.   Also, loopback.py didn't meet timing with the default pnr seed, but did meet timing with `--seed 198`.

I do have a working solution on Fomu with valentyusb and I think tinyusb, hooking into the uart / putc / printf stack (I hope that makes sense, I'm still figuring things out).   But I'd love to get to the same point with Luna too.   I'm not sure which if any pieces are missing.