greatscottgadgets/luna #85: Failure while following 'getting started' instructions

mubes, 2021-01-24:
Hi,
This one is probably sticky fingers, but following the instructions at https://luna.readthedocs.io/en/latest/getting_started.html at the point where the dry-run tests are conducted, I get the following error;

```
~/Builds/luna$ python3 applets/interactive-test.py --dry-run
WARNING | __init__    | Couldn't auto-detect connected platform. Assuming LUNA r0.2.
INFO    | __init__    | Building for LUNA r0.2...
Traceback (most recent call last):
  File "applets/interactive-test.py", line 400, in <module>
    tester = top_level_cli(InteractiveSelftest)
  File "/home/dmarples/Builds/luna/luna/__init__.py", line 144, in top_level_cli
    products = platform.build(fragment,
  File "/home/dmarples/Builds/luna/src/nmigen/nmigen/build/plat.py", line 95, in build
    plan = self.prepare(elaboratable, name, **kwargs)
  File "/home/dmarples/Builds/luna/src/nmigen/nmigen/build/plat.py", line 154, in prepare
    add_pin_fragment(pin, self.get_input_output(pin, port, attrs, invert))
  File "/home/dmarples/Builds/luna/src/nmigen/nmigen/vendor/lattice_ecp5.py", line 607, in get_input_output
    i_T=t[bit],
  File "/home/dmarples/Builds/luna/src/nmigen/nmigen/hdl/ast.py", line 261, in __getitem__
    raise IndexError(f"Index {key} is out of bounds for a {n}-bit value")
IndexError: Index 1 is out of bounds for a 1-bit value
~/Builds/luna$ 
```
...am I doing something dumb?

mubes, 2021-01-24:
actually, I seem to get this problem while trying to build any of the examples, e.g. `python3 acm_serial.py` or `python3 stress_test_device.py`.

mubes, 2021-01-24:
Problem seems to relate to recent changes in nmigen. Replicated by Greg Davill.

ktemkin, 2021-01-25:
Looks like this is a bug in current nmigen/master. I've submitted a PR over there fixing it: https://github.com/nmigen/nmigen/pull/582/files :)

mubes, 2021-01-25:
Yep, fixes it for me. Ta!
Not sure what your protocol is for ending issues? I'll leave it open in case you want to track-to-fix upstream.

closed 2021-02-03 by ktemkin