greatscottgadgets/luna #99: Component substitution question

pk-r7, 2021-03-16:
Trying to build a development version of this board, I noticed some of the components have very long leadtimes across the entire US supply. 
Specifically the FPGA and the oscillator.
The next step up in the ECP5 line is the LFE5U25. It is available and appears to be electrically compatible, I assume it would make a good replacement?
The entire line of SiTime oscillators is on a really long leadtime. I tried to compare footprints and specs and the closest I could find is a CMOS oscillator ASEM1-60.000MHZ-LC-T, however the ISA20-32AH-60.000MHZ is slightly larger and is an LVCMOS. Can anyone recommend a good replacement?

mossmann, 2021-03-16:
Yes, you can use LFE5U-25F, but I believe it will require one line to be changed in the gateware/platform file. Have you checked with Digi-Key? It appears that they have LFE5U-12F in stock, and they have SIT1602BC-23-33E-60.000000E as a "value added item" which I think always shows 0 stock even when they are able to fill orders.

I think the ASEM1-60.000MHZ-LC-T should work fine. However, we thought that the DSC60XX used prior to r0.3 would work, but it turned out to not be stable enough for reliable sustained USB transfers. The only oscillators we have thoroughly tested and approved are from the SiT1602B series.

If you haven't ordered PCBs yet, I suggest using the [r0.4 branch](https://github.com/mossmann/luna/tree/r0.4) (#97).

ktemkin, 2021-03-16:
> Yes, you can use LFE5U-25F, but I believe it will require one line to be changed in the gateware/platform file. 

If you have the board plugged in before you run the relevant nMigen script (which builds the gateware), the library will automatically detect the FPGA size and everything should work transparently. IIRC, you'll only have to tweak the `device` constant if you don't have the board plugged in when you build the gateware. :)

>  SIT1602BC-23-33E-60.000000E as a "value added item" which I think always shows 0 stock even when they are able to fill orders

The SiTime oscillators are programmable, so usually most vendors display themselves as having only the unprogrammed ones in stock. Some vendors (like DigiKey) will program them for you, but always show them as out of stock, since they make them on demand. Another option is to buy the unprogrammed oscillators and the programmer for them, which IIRC isn't all _that_ expensive.

pk-r7, 2021-03-16:
Thanks for the explanation on the oscillators. I tried to order them as a value added item, Digikey quoted me a leadtime of 24 to 26 weeks on the value added oscillator. Digi has the programmer at ~240, a bit much for a one off build. I just asked again if they would reprogram some for me. 

I did order off of the r0.4 branch from oshpark and oshstencils. Planned on sharing the project once I got them and confirmed the build.

mossmann, 2021-03-16:
Be aware that the OSH Park 4-layer stack-up differs from the stack-up specified for LUNA. I don't think it is likely to be an issue, but you _might_ see signal integrity problems using longer USB cables, for example.

pk-r7, 2021-03-16:
> Be aware that the OSH Park 4-layer stack-up differs from the stack-up specified for LUNA. I don't think it is likely to be an issue, but you _might_ see signal integrity problems using longer USB cables, for example.

More to learn, thanks for giving me a heads up on that.

pk-r7, 2021-04-09:
I discovered different issue during assembly. I can open a separate issue if you prefer.
U2 and U14 are listed as 74AUP2G126. I originally ordered X2SON units by using the 3dshape as a reference instead of what  I think is supposed to be a DFN1210-8 based off of the footprint. 

Assuming I have the package correct, may I suggest you change the reference in the BOM to 74AUP2G126RA3-7?

I ended up getting a different package with a slightly longer overall length but same pinout to attach I may or may not replace with the correct components once I get them. (Those package sizes are a nightmare to solder by hand) 
X2SON and XSON8 shown in the attached image (neither are the correct component).
[image]

mossmann, 2021-04-09:
> Assuming I have the package correct, may I suggest you change the reference in the BOM to 74AUP2G126RA3-7?

Oops! That is a bug. I intended to specify Diodes Inc. 74AUP2G126RA3-7.

gregdavill, 2021-04-23:
> I ended up getting a different package with a slightly longer overall length but same pinout to attach I may or may not replace with the correct components once I get them. (Those package sizes are a nightmare to solder by hand)

I also did the exact same thing, and ordered a part that's _slightly_ longer. I even checked and confirmed it wasn't the X2SON package, but didn't catch the slight pitch difference. :laughing: 
[image]

mossmann, 2021-08-30:
I've fixed this bug in #130 and will close this issue after r0.5 is merged.

mossmann, 2023-07-15:
closing as this was merged in #190 and has now migrated to https://github.com/greatscottgadgets/cynthion-hardware

closed 2023-07-15 by mossmann