greatscottgadgets/luna #105: Stream endpoint locks up

zyp, 2021-04-19:
Hi, I've got a `USBMultibyteStreamInEndpoint` that's being fed by a 16-byte wide stream with the `first` and `last` flags set, i.e. always sending 16B packets. This appears to work fine at lower data rates, but when I reach roughly 12k packets per second, it locks up in less than a minute.

Once it is locked up, the `ready` signal on the stream is stuck low, while polls to the endpoint are NAKed. Other functions appears to be unaffected and the device will happily reenumerate if I unplug it and plug it back in, but the endpoint remains stuck until I reset the FPGA.

Here's an excerpt from a bus capture:
[image]

zyp, 2021-04-19:
FWIW it fails in the same manner when downconverting the stream and feeding it to a regular `USBStreamInEndpoint`, so the issue is not limited to the `USBMultibyteStreamInEndpoint`.

ktemkin, 2021-04-20:
Interesting -- from our Discord exchange:

> the fact that it's NAK'ing, though, makes me suspect that maybe there's a logic bug in the state tracking that's making it think it's empty when it's not

Is there any chance you could hook up one of the LUNA ILAs and try to grab a capture of the various buffer pointers? A VCD of that might help a bunch with diagnosing things. :)

zyp, 2021-04-20:
I don't have access to hardware to test on right now, but I just eyeballed the code and have a theory about what is going on. Excerpt from [transfer.py](https://github.com/greatscottgadgets/luna/blob/main/luna/gateware/usb/usb2/transfer.py#L314):

```python3
                    # Otherwise, there's a possibility we already have a packet-worth of data waiting
                    # for us in our "write buffer", which we've been filling in the background.
                    # If this is the case, we'll flip which buffer we're working with, toggle our data pid,
                    # and then ready ourselves for transmit.
                    packet_completing = in_stream.valid & (write_fill_count + 1 == self._max_packet_size)
                    with m.Elif(~in_stream.ready | packet_completing):
                        m.next = "WAIT_TO_SEND"
                        m.d.usb += [
                            self.data_pid[0]   .eq(~self.data_pid[0]),
                            read_stream_ended  .eq(0)
                        ]

                    # If neither of the above conditions are true; we now don't have enough data to send.
                    # We'll wait for enough data to transmit.
                    with m.Else():
                        m.next = "WAIT_FOR_DATA"
```

`packet_completing` doesn't account for the possibility of it being a short packet, which means that if we get the ACK in the same cycle as the last byte of the next packet, we'll move back to `WAIT_FOR_DATA` rather than `WAIT_TO_SEND`. Since the buffers are now full, `WAIT_FOR_DATA` is never getting `will_end_packet` and therefore remains stuck there.

I believe adding a check for `in_stream.last` to `packet_completing` will fix the problem and will test this later tonight.

zyp, 2021-04-20:
I just ran tests with and without the patch in #106.

Without the patch it locked up in 13 seconds. With the patch it ran for over ten minutes without any issues.

PR #106 (2021-04-26): gateware.transfer: Fix corner case causing a potential lockup for short packets.
A packet is complete either when it has reached the max packet size or the last flag is set.
We need to check both.

Fixes #105
commit f713312f (2021-04-26): gateware.transfer: Fix corner case causing a potential lockup for short packets.

A packet is complete either when it has reached the max packet size or the last flag is set.
We need to check both.

Fixes #105
closed 2021-04-26 by ktemkin