greatscottgadgets/luna #116: USBDescriptorStreamGenerator expects StandardDescriptorNumbers?

puckipedia, 2021-05-07:
When elaborating any of the USB examples (for the ULX3S, at least) the USB code seems to fail with:

```
  File "[..]/LUNA/luna/gateware/usb/usb2/descriptor.py", line 113, in elaborate
    setattr(m.submodules, f'USBDescriptorStreamGenerator({type_number.name},{index})', generator)
AttributeError: 'int' object has no attribute 'name'
```

To me it looks like the `StandardDescriptorNumbers(IntEnum)` is converted to a byte and then to an integer:
https://github.com/usb-tools/python-usb-protocol/blob/a8ff3d90cc103b8fee482f956298fe86bc76e255/usb_protocol/types/descriptor.py#L116-L120

Thus losing its name attribute, and breaking elaboration. If I manually remove the `.name` or construct the `IntEnum` back, it seems to work fine.

zyp, 2021-05-07:
I can confirm the issue. The name lookup was added in #113 as part of adding the unit tests, presumably to make the test output more readable.

@twam How do we fix this? Do we remove the name lookup again, or do we ensure that the name is always available?

twam, 2021-05-07:
The nice name was only added for readability of the tests. :)

I'd suggest something like
```    setattr(m.submodules, f'USBDescriptorStreamGenerator({getattr(type_number, 'name', type_number)},{index})', generator)```

I can create a PR and a test-case for it tonight.

PR #117 (2021-06-03): Only use .name property if type is of StandardDescriptorNumbers
Fixes #116.
commit 1d9f3733 (2021-06-03): Only use .name property if type is of StandardDescriptorNumbers (#117)

* Only use .name property if type is of StandardDescriptorNumbers

Also added a non-standard descriptor to the unit tests which
tests the issue.

Fixes #116.

* Use isinstance() instead of type()
closed 2021-06-03 by ktemkin