greatscottgadgets/luna #123: Trying to run interactive-test.py on r0.2 luna board

tcal-x, 2021-07-16:
There's a non-zero chance of user error here.   This is what I attempted (from memory):

* I created and entered a virtual environment (python -m venv)
* In nmigen/nmigen/, I ran `python3 ./setup.py install`
* In greatscottgadgets/luna/, I ran `pip3 install poetry`, `poetry install`
* I connected a cable from my laptop's USB port to the Luna board's "Debug/Alt Port"
  * There was a slow-blinking green LED, and a row of dim red (w/ one green) LEDs
* I ran `poetry run applets/interactive-test.py`
  * The green LED(s) momentarily ran a bouncing back and forth pattern
  * The row of red LEDs got brighter
  * This output was returned on the terminal:
```
INFO    | __init__    | Building and uploading gateware to attached LUNA r0.2...
INFO    | __init__    | Upload complete.
INFO    | selftest    | Connected to onboard debugger; hardware revision r0.2 (s/n: eebbd5a905b435050213e29382f031ff).
INFO    | selftest    | Running tests...

Automated tests:
        Debug module:   ✗ FAILED        (register 1 was 63, not expected 1413829460)
        Host PHY:       ✗ FAILED        (PHY register 0 was 3182280431, not expected 36)
        HyperRAM:       ✗ FAILED        (RAM register 0 was 3182280431, not one of expected (3201, 3206))
        Sideband PHY:   ✗ FAILED        (PHY register 0 was 3182280431, not expected 36)
        Target PHY:     ✗ FAILED        (PHY register 0 was 3182280431, not expected 36)
```

FFY00, 2021-07-16:
FYI me and @perigoso were seeing similar, if not the same, failures due to the oscillator, which is quite finicky.

tcal-x, 2021-07-16:
> FYI me and @perigoso were seeing similar, if not the same, failures due to the oscillator, which is quite finicky.

Thanks, that's useful info!   Is there any known workaround?

FFY00, 2021-07-16:
See https://github.com/greatscottgadgets/luna/wiki/Hardware-Changes#revision-02-1. If you are not using the new oscillator, that'd be a starting point. We were not able to get the recommended oscillator where we live, so we had to order equivalents, the ones I ordered did not work, the ones @perigoso ordered did. If you need, we could dig up the part number for you.

tcal-x, 2021-07-16:
Thanks @FFY00 ; it's probably beyond my ability to replace a chip on the board.    I swear I saw some images calling out the parts on the Luna board (not sure if r0.2 or the new one or both), but now I can't find them.

closed 2021-07-16 by tcal-x