greatscottgadgets/luna #124: Future of LUNA?

nekrondev, 2021-07-27:
I'm not quite sure whatever reason has led for @ktemkin to retire from the LUNA project and to resign from GSG, but I'm wondering how the former head of software development and main contributor to LUNA project and it's USB tools will be substituted in the future?

I've backed LUNA H/W at Crowdsupply because I've got deep respect for the work of Kate regarding this project and her attitude in general so this was the main reason I voted to support GSG and her work.

Reading about the break up just 24h's after the campaign had started was a hit in the face for me and somehow found betrayed.
Even the best H/W that for sure GSG will deliver next year is worthless if the software development has stalled. 

Please, a clear statement from @mossmann or whoever is responsible for this farce would be very appreciated.

mossmann, 2021-07-27:
While Kate was a very important part of the team and her contributions will be missed, LUNA is a team project. We are currently refocusing our remaining team members on LUNA to ensure the project gets proper support, and we will be hiring soon to keep the project on track for release. Already on our team is @miek who has developed hardware, gateware, and software for Amalthea, a project based on LUNA, and I took over LUNA hardware design work last year. Please be assured that software development efforts will resume and that LUNA will ship as planned with all capabilities listed on Crowd Supply.

ktemkin, 2021-07-27:
Personally, I have a lot of respect for @miek's abilities -- he's worked on LUNA-like projects in the past, and has been a contributor to a variety of nMigen-language projects. While I can't speak for GSG or their plans, I do have a lot of faith in @miek's ability to get the gateware/software side of things done.

closed 2021-07-28 by mossmann

nekrondev, 2021-07-28:
@mossmann: Thanks for clarification of the current situation and how it will evolve in the near future. I personally have high hopes getting a smashing brilliant USB tinkering device based on the fundamental work of Kate and Mike's further improvements.

@ktemkin  @miek: Good luck and all the best to current and former LUNA project members.