greatscottgadgets/luna #129: Case silkscreen feedback

cyrozap, 2021-08-12:
Yesterday, [a photo of the case](https://cdn.discordapp.com/attachments/816773068277809182/875107561060327525/image0.jpg) that was posted in the Discord channel. I was told to post my feedback regarding the silkscreen design here, so I'm doing that now.

I'll preface these statements by saying that I'm not a designer and don't have any experience in that area--these are just the issues that I, as a (future) user, had with the design. So, please don't grant these statements any undue authority. Also, these are just my opinions, so please feel free to ignore them if you think they're wrong.

# Issues

## Text orientation

My first issue is that the text is oriented in three directions, and somewhat inconsistently so. For instance, the "Host" and "Sideband" labels are rotated 90 degrees clockwise, and the "Target" label is rotated 90 degrees in the opposite direction. That all makes sense, since it means you can easily read the labels while looking at the ports you're plugging cables into. But then the labels for the "DFU" and "Reset" buttons have no rotation, despite the buttons being on the same sides of the case as those ports whose labels are rotated 90 degrees. I can understand the thought process behind that--the expectation is that the device would primarily be oriented with the ports on the left and right sides, and the user can easily read the "DFU" and "Reset" labels in that orientation and then just feel for the buttons. But to have text that's so close together oriented perpendicularly, as in the case of the "Sideband" and "DFU" labels, is somewhat confusing, and doesn't look very good. But then rotating "DFU" to be in line with "Host and "Sideband" would put it in conflict with the "Debug" LED labels, so something would need to be done about those, too.

## Indicators for each port's size

Without port width indicators like the "Target" label has (the lines the the left and right of the text), it's non-obvious where each port starts and ends. e.g., it'd be nice to see something like `┌──Host──┐┌Sideband┐` (or `┌──Host──┬Sideband┐` or whatever) to make that more clear.

## LED labeling

Similarly, the "Debug" and "FPGA" labels could use some separation from and better indication of what LEDs they cover. For instance, maybe the LED spots could have a line covering all them, and then put the "Debug"/"FPGA" labels above that line. Maybe something like this (but with the text actually underlined as well):

```
__Debug__   ___FPGA____
● ● ● ● ●   ● ● ● ● ● ●
A B C D E   5 4 3 2 1 0
```

Or, if labels start to get cluttered, the LED labels could be put in a "call out" (I don't know if that's the right term), where a line goes over the top of the LED spaces and links to a different area where the actual labels are. Kind of like what you might see on some PCBs with densely-packed clusters of components.

## Clutter

The design is simultaneously "too cluttered" and "has too much negative space". What I mean by that is, first, it seems the logos for the Luna itself and GSG were placed to fill as much of the space as they could, while leaving a large empty space in the middle. That's not to say I'd like to see that space filled--rather, I think it would be better to decrease the sizes of both logos, and also possibly even condense them into one (GSG logo plus "LUNA" text, and nothing else) or put them on the back of the case. Speaking of which...

The CE mark should not be on the front of the case. I realize that it was probably placed there since it may cost less to only mark the top side of the case, but whether or not something is CE-certified is generally not something I need to stare at every day. Instead, what I as a user find useful to see are the following (in no particular order):

- The port labels (so I know what to plug where, without having to consult the manual).
- The LED labels (again, so I don't have to consult the manual to know what they mean).

So if it's at all possible, I would suggest putting the full Luna logo and the CE mark on the back of the case, and then, since the GSG name is already a part of the Luna logo, skip the GSG gears entirely. Or, the GSG logo could be kept on the front side, but centered.

# Design mockup

A picture's worth a thousand words, so here's a sloppy mockup I did of a "fixed" design:

[image]

The two big circles are where the GSG logo would go. The design is not to scale (again, it's a sloppy mockup), and so maybe some of these design decisions might not work exactly, but I think this does a decent job illustrating my suggestions. I'm still not really happy with the labeling for the FPGA LEDs, since having all the digits in a horizontal line makes me want to read them like a single number, but I can't really think of a much better way to do that that wouldn't end up making some other aspect of the design worse.

Anyways, I'm eager to hear what people think about my comments/mockup.

perigoso, 2021-08-12:
I like these suggestions

A tiny detail I'd also change is the reset label, from "RESET" to just "RST", makes it more symetric with DFU and since the button itself occupies a small area it helps pinpoint it

straithe, 2021-08-13:
I really like your proposal for labelling the LEDs!

eduncan911, 2021-09-06:
I used the CrowdSupply contact page to ask them about my purchase and getting a blank case instead.  This was the reply:

> Thank you for your question and suggestion! We will not be doing a run of blank LUNA cases, but we will eventually be releasing design files so you can 3D print your own case, if you desire.
> 
> BP

closed 2021-10-15 by straithe