greatscottgadgets/luna #132: pyproject.toml points to a broken apollo-fpga dependency

supersat, 2021-09-15:
pyproject.toml currently specifies the apollo-fpga package as a pypi dependency with version ^0.0.4. This has a few issues:

- Since 4 is the only non-zero number in the version, AFAIK this keeps the dependency stuck at version 0.0.4
- The v0.0.4 apollo-fpga package in pypi does not actually match the state of the repo when its version was bumped to 0.0.4 (the flash commands are missing)
- The apollo-fpga pypi package is still maintained by Kate, who is presumably busy with other things

Other dependencies are specified based on their git URL. Perhaps that would be appropriate here?

mossmann, 2021-09-27:
Kate gave me control of the pypi packages.

martinling, 2024-06-21:
We're in sunny 1.0.x land now.

closed 2024-06-21 by martinling