greatscottgadgets/luna #163: ECP5 protocol error for descriptors longer than 128 bytes

BrettRD, 2022-03-23:
I'm porting a bit of gateware from a Deca to a Colorlight, and I've run into a bug when using `GetDescriptorHandlerBlock` which is not encountered with `GetDescriptorHandlerDistributed`
Exploration here: https://github.com/amaranth-community-unofficial/deca-usb2-audio-interface/issues/5

If the descriptor grows to form a URB greater than 128 Bytes using `GetDescriptorHandlerBlock` it wraps and truncates the descriptor.

If I set `LUNA_AVOID_BLOCKRAM`, it uses a `GetDescriptorHandlerDistributed` instead and happily enumerates with a descriptor URB of 218 Bytes (packet size of 282 bytes)

This flag affects exactly one condition in Luna here: https://github.com/greatscottgadgets/luna/blob/08b035c9c2b053d7edffb0d220d948b5c2ca927e/luna/gateware/usb/request/standard.py#L129-L132

These two descriptor handlers ought to have identical behaviour aside from timing constraints and resource usage

This bug seems to affect the ECP5 platform and not the DECA platform, so I suspect it's an issue related to platform-specific block-ram features and allocation

jboone, 2022-05-02:
I don't know how useful this data point is, but I've had LUNA USB enumeration problems, seemingly at random, which were magnified by adding a bunch of USB UAC descriptors from a [DECA-based audio project](https://github.com/amaranth-community-unofficial/deca-usb2-audio-interface/blob/main/gateware/deca_usb2_audio_interface.py). If I remove only those descriptors (but leave the ones for the non-audio portion of my project), the generated bitstreams are reliable. If I add back the audio project's descriptors (which are many!), the problem returns. If I use `LUNA_AVOID_BLOCKRAM` as suggested, the project with the audio descriptors seems so far to be entirely reliable. So I'm guessing there's some sort of block RAM timing closure or mischaracterization issue in prjtrellis? My USB clock timing winds up clearing the 60MHz requirement by more than 10MHz, so nextpnr *thinks* everything's cool. But maybe not...

BrettRD, 2024-11-21:
My apologies for the noise, this was a hardware bug with my ULPI device.

I'm using the Waveshare USB3300 which generates its own clock, and has a connector that makes it hard to have short trace lengths.  The FPGA is clocked from the ULPI so propagation delay counts double.
By *extending* the ULPI cable to about 200mm of total trace length and *inverting* the fpga clock input pin, I have made a cursed contraption that enumerates with long descriptors.

There is a timing (or perhaps glitch) difference between Block and Distributed handlers, but it's practically cosmetic.

closed 2024-11-21 by BrettRD