greatscottgadgets/luna #168: Cannot build Analyzer for luna.gateware.platform.lambdaconcept:USB2SnifferPlatform due to missing power_a_port

whinis, 2022-05-07:
Whenever I attempt to build for the USB2Sniffer I get the following error 
`amaranth.build.res.ResourceError: Resource power_a_port#0 does not exist`
line 102 of the analyzer.py also suggest more issues due to 
```       
m.d.comb += [
            platform.request("power_a_port").o      .eq(0),
            platform.request("pass_through_vbus").o .eq(1),
        ]
```

martinling, 2022-05-09:
Those are signals that exist on the LUNA hardware platform, which tell it to let power pass between the target A and C ports and not supply its own power to the A port.

As I understand it, the USB2Sniffer hardware is always in that configuration, so as a quick fix you should be able to just delete that assignment block - though obviously it would be nicer to have a patch that handled this properly for the different platforms.

martinling, 2024-02-13:
We've moved the analyzer gateware into the [Cynthion](https://github.com/greatscottgadgets/cynthion) repository as per #232. As such I'm going to close this issue, since the code is no longer part of the LUNA repo.

If folks want to port it to other platforms they're welcome to, but our development is going to be focused on Cynthion.

closed 2024-02-13 by martinling