greatscottgadgets/luna #185: design r0.6

mossmann, 2022-11-29:
- [ ] change IC1 (FPGA) from 256 caBGA to 381 caBGA package, retaining ability to switch back to 256 caBGA later
- [ ] change U8, U9, U11 (USB PHYs) to USB3320
- [ ] rename Sideband port to "Control" and use as primary port for control of both FPGA and Apollo)
- [ ] rename Host port to "Aux" for auxiliary functions including MitM
- [ ] enhance Target-C PD/CC I/O with FUSB302B or similar Type-C controller
- [ ] connect shields of all USB connectors together
- [ ] disconnect USB shields from GND except through unpopulated ferrites
- [ ] replace U1, U12, U13 (Target VBUS switches) with FET solution
- [ ] add Target VBUS voltage and current monitoring
- [ ] replace d1, D8 (5V diode OR) with FET solution
- [ ] move HyperRAM to FPGA pins in a DQS group if possible for both 381 caBGA and 256 caBGA variants
- [ ] increase clearance around U6 (microcontroller) to allow future replacement if necessary
- [ ] restore 1.8 V supply as an assembly option, supporting either 1.8 V or 3.3 V HyperRAM
- [ ] add pin straps for hardware version detection
- [ ] increase LED spacing
- [ ] add USB switch to Control port
- [ ] improve microcontroller D+/D- routing/reference plane
- [ ] maybe remove J5 (FPGA JTAG) if it is in the way
- [ ] maybe move Pmod connectors to the same side to support 24-pin Pmods
- [ ] maybe remove unpopulated SMA connectors
- [ ] maybe replace SMA connectors with high speed edge connector

mossmann, 2023-02-07:
Work in progress is here: https://github.com/mossmann/luna/tree/hw-r0.6

yyrfejx884, 2023-03-24:
Can you add pin headers to target d- and d+ lines?

yyrfejx884, 2023-03-24:
There are some cases when you need to put some extra circuit on these lines temporarily for example: https://www.huaweinewos.com/huawei-harmonyos-support-harmony-tp-cable.html
In this case you need to put two pull-ups on the data lines till the fastboot loads then you can remove them and use the usb as normal.

mossmann, 2023-03-27:
> There are some cases when you need to put some extra circuit on these lines temporarily for example: https://www.huaweinewos.com/huawei-harmonyos-support-harmony-tp-cable.html In this case you need [...]

We won't be adding a D+/D- pin header in r0.6 for the signal integrity reason @martinling mentioned and also due to lack of board area and time (we have already had r0.6 prototype PCBs made). Harmony TP interfacing will have to be done with an external cable or by soldering resistors to the through-hole TARGET A pins.

If you implement the USB communication with Cynthion (not a passthrough host), you could use the additional TARGET port as a place to plug in a small resistor circuit instead of using a special cable.

mossmann, 2023-07-15:
closing as this (and more) has been merged to: https://github.com/greatscottgadgets/cynthion-hardware

closed 2023-07-15 by mossmann