greatscottgadgets/luna #187: tinyusb_soc.py - ValueError: Memory map has been frozen

gaudat, 2022-12-21:
I am trying to use tinyusb_soc.py to evaluate this core. That part of the repository seems to be unmaintained but I had to try it as I could not find any other examples.

I am greeted with this exception on line 87. The memory map of SimpleSoC seems to be already fixed when returning from the constructor. 

Should I dig further into this path? Or is there other documentation or examples for evaluating Luna as a SoC peripheral?

gaudat, 2022-12-21:
In latest Amaranth assigning a memory map to a bus freezes the memory map preventing any modification afterwards.

Fix: `luna/gateware/soc/memory.py` line 88 

        self.bus = wishbone.Interface(addr_width=self.local_addr_width, data_width=data_width, granularity=granularity)
        self.bus.memory_map = memory.MemoryMap(addr_width=self.bus_addr_width, data_width=granularity)
        self.bus.memory_map.add_resource(self, size=2 ** addr_width)

becomes:

        self.bus = wishbone.Interface(addr_width=self.local_addr_width, data_width=data_width, granularity=granularity)
        memory_map = memory.MemoryMap(addr_width=self.bus_addr_width, data_width=granularity)
        memory_map.add_resource(self, size=2 ** addr_width)
        self.bus.memory_map = memory_map

antoinevg, 2022-12-22:
Thank you for reporting this!

We have an open PR bringing SimpleSoC up to date that is waiting on review: 

https://github.com/greatscottgadgets/luna/pull/188

gaudat, 2022-12-23:
Thank you for updating the SimpleSoC core! Looking forward to it.

closed 2023-01-02 by miek