greatscottgadgets/luna #194: Cannot install from git

jeremyherbert, 2023-02-10:
Hi,

When trying to install from git as per the documentation, I received the following error:

```
Creating virtualenv luna-MtvwdPsJ-py3.10 in /home/jeremy/.cache/pypoetry/virtualenvs
Installing dependencies from lock file

Because lambdasoc (0.1.dev27+gefd5442) @ git+https://github.com/ktemkin/lambdasoc.git@HEAD depends on nmigen (*) which doesn't match any versions, lambdasoc is forbidden.
So, because luna depends on lambdasoc (0.1.dev27+gefd5442) @ git+https://github.com/ktemkin/lambdasoc.git, version solving failed.
```

I saw in another issue I could instead try `pip install .`, but this also gives an error:

```
Defaulting to user installation because normal site-packages is not writeable
Processing /home/jeremy/Downloads/luna
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting amaranth-boards@ git+https://github.com/amaranth-lang/amaranth-boards.git@main
  Cloning https://github.com/amaranth-lang/amaranth-boards.git (to revision main) to /tmp/pip-install-r5euf53l/amaranth-boards_5e1919b96376435392db67a27f19d5e4
  Running command git clone --filter=blob:none --quiet https://github.com/amaranth-lang/amaranth-boards.git /tmp/pip-install-r5euf53l/amaranth-boards_5e1919b96376435392db67a27f19d5e4
  Resolved https://github.com/amaranth-lang/amaranth-boards.git to commit 609a65c163c1374fb5f9c528253c36caa8edd56f
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package amaranth-boards produced metadata for project name unknown. Fix your #egg=amaranth-boards fragments.
Discarding git+https://github.com/amaranth-lang/amaranth-boards.git@main: Requested unknown from git+https://github.com/amaranth-lang/amaranth-boards.git@main (from luna==0.1.0.dev0) has inconsistent name: filename has 'amaranth-boards', but metadata has 'unknown'
Collecting usb-protocol@ git+https://github.com/usb-tools/python-usb-protocol
  Cloning https://github.com/usb-tools/python-usb-protocol to /tmp/pip-install-r5euf53l/usb-protocol_867f717767d54f2c9f5da760f02d3564
  Running command git clone --filter=blob:none --quiet https://github.com/usb-tools/python-usb-protocol /tmp/pip-install-r5euf53l/usb-protocol_867f717767d54f2c9f5da760f02d3564
  Resolved https://github.com/usb-tools/python-usb-protocol to commit a9830731a8de6946bfee014e6b0188ee83a716f2
  Preparing metadata (setup.py) ... done
Collecting amaranth@ git+https://github.com/amaranth-lang/amaranth.git@main
  Cloning https://github.com/amaranth-lang/amaranth.git (to revision main) to /tmp/pip-install-r5euf53l/amaranth_f62391ffd97b4bc685174083be7a2465
  Running command git clone --filter=blob:none --quiet https://github.com/amaranth-lang/amaranth.git /tmp/pip-install-r5euf53l/amaranth_f62391ffd97b4bc685174083be7a2465
  Resolved https://github.com/amaranth-lang/amaranth.git to commit 666ee27fd0fa9fb90109e0e201f878f9ab034c85
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package amaranth produced metadata for project name unknown. Fix your #egg=amaranth fragments.
Discarding git+https://github.com/amaranth-lang/amaranth.git@main: Requested unknown from git+https://github.com/amaranth-lang/amaranth.git@main (from luna==0.1.0.dev0) has inconsistent name: filename has 'amaranth', but metadata has 'unknown'
Collecting amaranth-soc@ git+https://github.com/amaranth-lang/amaranth-soc.git@main
  Cloning https://github.com/amaranth-lang/amaranth-soc.git (to revision main) to /tmp/pip-install-r5euf53l/amaranth-soc_a01fc66bb4ce41f1a7613f4e087ec057
  Running command git clone --filter=blob:none --quiet https://github.com/amaranth-lang/amaranth-soc.git /tmp/pip-install-r5euf53l/amaranth-soc_a01fc66bb4ce41f1a7613f4e087ec057
  Resolved https://github.com/amaranth-lang/amaranth-soc.git to commit bddf5a82ff63ad55c13138a654da511317f6861c
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package amaranth-soc produced metadata for project name unknown. Fix your #egg=amaranth-soc fragments.
Discarding git+https://github.com/amaranth-lang/amaranth-soc.git@main: Requested unknown from git+https://github.com/amaranth-lang/amaranth-soc.git@main (from luna==0.1.0.dev0) has inconsistent name: filename has 'amaranth-soc', but metadata has 'unknown'
Requirement already satisfied: pyvcd<0.3.0,>=0.2.4 in /home/jeremy/.local/lib/python3.10/site-packages (from luna==0.1.0.dev0) (0.2.4)
Collecting ziglang<0.9.0,>=0.8.0
  Downloading ziglang-0.8.1-py3-none-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (68.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 68.1/68.1 MB 1.9 MB/s eta 0:00:00
Collecting pyusb<2.0.0,>=1.1.1
  Downloading pyusb-1.2.1-py3-none-any.whl (58 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.4/58.4 KB 23.5 MB/s eta 0:00:00
Collecting libusb1<2.0.0,>=1.9.2
  Downloading libusb1-1.9.3-py3-none-any.whl (60 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 60.5/60.5 KB 28.7 MB/s eta 0:00:00
ERROR: Could not find a version that satisfies the requirement amaranth-boards (unavailable) (from luna) (from versions: 0)
ERROR: No matching distribution found for amaranth-boards (unavailable)
```

Could you please help me understand what the correct way to install this package is?

antoinevg, 2023-02-10:
The plan moving forward is to stop using poetry entirely in favour of pip.

`pip install .` _should_ be working at the moment against HEAD but it's good to know when it isn't!

Can you please provide a bit more info about your environment:

* OS & version
* Python version
* Are you using a virtual environment tool such as `virtualenv` or `pyenv` ?

Environments I've tested and can confirm work from a fresh environment:

* macOS 12 & 13, Ubuntu 22.04, Arch latest
* Python 3.10 and 3.11, PyPy 3.9

jeremyherbert, 2023-02-10:
Hi,

Sorry, I should have provided that info earlier. I’m using Ubuntu 22.04 with python from the apt repo, 3.10. The only other things installed via pip are litex and amaranth (and their dependencies). No virtual environment, I’m running a separate VM with the fresh Ubuntu install so there isn’t really a need for one.

antoinevg, 2023-02-10:
Ah okay, this could indeed be related to the recent shift amaranth made to `pyproject.toml`.

There's an open PR which is related:

https://github.com/greatscottgadgets/luna/pull/192/files#diff-4d7c51b1efe9043e44439a949dfd92e5827321b34082903477fd04876edb7552

In the meantime you could try this change on your local `requirements.txt`:

```diff
- -e git+https://github.com/amaranth-lang/amaranth.git#egg=amaranth
+ amaranth -e git+https://github.com/amaranth-lang/amaranth.git#egg=amaranth
```

jeremyherbert, 2023-02-11:
Seems that this problem is related to pip 22 which ships with ubuntu by default. Upgrading to pip 23 and using the PR you mentioned works fine.

closed 2023-02-11 by jeremyherbert