greatscottgadgets/luna #229: can't install luna in ubuntu 22.04

matisTopPath, 2024-01-03:
Hi, I try to install luna to Ubuntu 22.04 and run applets/interactive-test.py
But always show errors.

Would someone be able to give the environment lib list?
I have read the install documents, but I am not sure what's missing.

thanks alot.

miek, 2024-01-04:
Could you post the log of errors that you're seeing?

matisTopPath, 2024-01-05:
@miek, the error log,  I try to install luna on pi4 and ubuntu 22.04 but it always shows this error. : (

dev@dev:~/luna $ sudo python3 applets/interactive-test.py 
`
Traceback (most recent call last):
  File "/home/dev/luna/applets/interactive-test.py", line 291, in <module>
    tester = top_level_cli(InteractiveSelftest)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/luna/__init__.py", line 93, in top_level_cli
    platform = get_appropriate_platform()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/luna/gateware/platform/__init__.py", line 100, in get_appropriate_platform
    platform = PLATFORM_FOR_REVISION[version]()
               ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
KeyError: (1, 3)`

martinling, 2024-06-21:
This was happening because there wasn't a platform definition in the package for r1.3 boards.

Several things have changed since this issue was raised:
- The LUNA hardware design has been [renamed to Cynthion](https://www.crowdsupply.com/great-scott-gadgets/luna/updates/renaming-luna-hardware-to-cynthion).
- Platform definitions are no longer part of the LUNA package, they're maintained in other packages/repositories.
- Platforms for Cynthion r1.3 and all other board versions are now in the [cynthion](https://github.com/greatscottgadgets/cynthion) package.

If you install the `cynthion` package from the repository linked above, everything should work now.

closed 2024-06-21 by martinling