greatscottgadgets/luna #243: what the different repo luna and cynthion

imatespl, 2024-04-11:
hi, i can not find doc description what different repo of [luna](https://github.com/greatscottgadgets/luna) and [cynthion](https://github.com/greatscottgadgets/cynthion)?
and i read the [packetry](https://github.com/greatscottgadgets/packetry) doc, there is a [LUNA](https://github.com/greatscottgadgets/luna) USB analyzer applet，but i can not find the applet gateware.
and there also a repo [ViewSB](https://github.com/greatscottgadgets/ViewSB), it is a usb analyzer soft like packetry, which should use?

miek, 2024-04-11:
The hardware device that was previously called LUNA was renamed to Cynthion, and the software specific to that was moved into the cynthion repo. The analyzer applet can be found there: https://github.com/greatscottgadgets/cynthion/tree/main/cynthion/python/src/gateware/analyzer. This repo (luna) is now just for the gateware framework for building USB devices.

ViewSB is also USB analysis software, but it was replaced by Packetry and that's what we'll be developing going forward, so I would recommend using that.

imatespl, 2024-04-11:
Thank you very much, i follow the [ Cynthion](https://github.com/greatscottgadgets/cynthion) BUILD.md
``` bash
cd cynthion.git/cynthion/python/
LUNA_PLATFORM="cynthion.gateware.platform:CynthionPlatformRev1D3" make analyzer
```
1 question
This will compile get the analyzer gateware and can flush to  Cynthion use [apollo](https://github.com/greatscottgadgets/apollo) right?
there a   Moondancer soc gateware and firmware, no need for cynthion when use as a analyzer right?
just flush the analyzer gateware and use Packetry will can capture the usb packet right?
2 question
The Packetry can capture usb packet in background no ui mode?

imatespl, 2024-04-11:
i find the [README.md](https://github.com/greatscottgadgets/cynthion/blob/main/cynthion/python/src/gateware/analyzer/README.md), now  the 1 question is sure for me，just confirm 2 question packetry can capture the usb packet in background like tcpdump use usbmon, Thanks!

imatespl, 2024-04-12:
i compile the packetry, it's can not use in background like tcpdump, close this issue

closed 2024-04-12 by imatespl