greatscottgadgets/luna #270: AttributeError: 'Signal' object has no attribute 'o'

retif, 2024-07-05:
code version: a87888ea72f806f05660aa1a4dd336fc5d80a6f2

```
Traceback (most recent call last):
  File "/home/oleks/projects/orbtrace/./orbtrace_builder.py", line 112, in <module>
    main()
  File "/home/oleks/projects/orbtrace/./orbtrace_builder.py", line 101, in main
    builder.build(**trellis_argdict(args), run=args.build)
  File "/home/oleks/projects/litex/litex/litex/soc/integration/builder.py", line 373, in build
    self.soc.finalize()
  File "/home/oleks/projects/litex/litex/litex/soc/integration/soc.py", line 1477, in finalize
    Module.finalize(self)
  File "/home/oleks/projects/litex/migen/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oleks/projects/litex/migen/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
                    ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oleks/projects/litex/migen/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/home/oleks/projects/litex/migen/migen/fhdl/module.py", line 157, in finalize
    self.do_finalize(*args, **kwargs)
  File "/home/oleks/projects/orbtrace/orbtrace/amaranth_glue/wrapper.py", line 84, in do_finalize
    f.write(self.generate_verilog())
            ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oleks/projects/orbtrace/orbtrace/amaranth_glue/wrapper.py", line 61, in generate_verilog
    fragment = ir.Fragment.get(self.m, None).prepare(ports = ports)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oleks/toolchains/oss-cad-suite-20240705/lib/python3.11/site-packages/amaranth/hdl/ir.py", line 42, in get
    new_obj = obj.elaborate(platform)
              ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oleks/toolchains/oss-cad-suite-20240705/lib/python3.11/site-packages/amaranth/hdl/dsl.py", line 562, in elaborate
    fragment.add_subfragment(Fragment.get(submodule, platform), None)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oleks/toolchains/oss-cad-suite-20240705/lib/python3.11/site-packages/amaranth/hdl/ir.py", line 42, in get
    new_obj = obj.elaborate(platform)
              ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oleks/toolchains/oss-cad-suite-20240705/lib/python3.11/site-packages/amaranth/hdl/dsl.py", line 560, in elaborate
    fragment.add_subfragment(Fragment.get(self._named_submodules[name], platform), name)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oleks/toolchains/oss-cad-suite-20240705/lib/python3.11/site-packages/amaranth/hdl/ir.py", line 42, in get
    new_obj = obj.elaborate(platform)
              ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oleks/toolchains/oss-cad-suite-20240705/lib/python3.11/site-packages/luna/gateware/interface/ulpi.py", line 857, in elaborate
    m.d.comb += self.ulpi.rst.o.eq(ResetSignal(raw_clock_domain)),
                ^^^^^^^^^^^^^^^
AttributeError: 'Signal' object has no attribute 'o'
```

What this error could mean?

martinling, 2024-07-05:
This is to do with some changes in Amaranth and LUNA.

There was a lot of code that used to pass around `Pin` objects (which have `.o` for output signal and other attributes), which has been changed to pass `Signal` objects instead, to maintain compatibility with newer Amaranth versions.

The main change on the LUNA side was #219.

However, there's still some places where things aren't quite right, as I ran into in https://github.com/greatscottgadgets/cynthion/pull/101.

This needs some further investigation.

miek, 2024-07-05:
While it's similar to that issue I'm not sure that's what's going on here. It looks like you're using the version of Amaranth included in the OSS CAD suite, which will be 0.5.0 or newer and is not yet supported by LUNA. Also the LUNA tag you've picked is very old and doesn't work with any recent version of Amaranth.

None of this should be an issue though, since orbtrace includes all the correct versions of the dependencies it needs and it shouldn't be pulling from other paths. If you're building orbtrace from git, you need to be careful to check out the submodules too (with `git submodule update --init --recursive`).

retif, 2024-07-05:
@miek you absolutely right, thanks! After making `git submodule update` it works ok, thank you very much for pointing out.

@martinling sorry for bothering, closing the issue

closed 2024-07-05 by retif