greatscottgadgets/luna #284: LUNA not compatible with Yosys 0.48

shareefj, 2024-12-29:
Yosys 0.48 seems to have removed `read_ilang` so installing the latest of all dependencies doesn't work.  Perhaps you could pin the known good versions of all dependencies.

shareefj, 2024-12-29:
Meh, opened on wrong project.  Should be Amaranth...

miek, 2025-01-02:
I think it's still relevant here. Amaranth released a fix in v0.5.4, but we don't support the 0.5.x branch yet. One option would be to see if they'll backport to 0.4.x too, but I think I will just get on with finishing https://github.com/greatscottgadgets/luna/pull/278 and then we can pin the fixed 0.5 version.

antoinevg, 2025-01-06:
For now, the most recent Yosys release that works with Amaranth 0.4.x is:

https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2024-11-01